KumarRobotics/kr_mav_control #135: Change from wercker CI to github actions

kartikmohta, 2021-02-08:
Wercker doesn't seem to be working well these days, and since this is now open-source, better to just use github actions.

tdinesh, 2021-02-08:
Sounds good to me

tdinesh, 2021-02-09:
Lets use [ROS CI action](https://github.com/marketplace/actions/ros-2-ci-action)?

kartikmohta, 2021-02-09:
It looks ok but requires specifying each package individually in the `package-name` option, not ideal for our case.

tdinesh, 2021-02-09:
Any other examples that you know of? Else, have to start from scratch.

kartikmohta, 2021-02-09:
All the steps required are in the `wercker.yml` already, could it be so hard to do our own? (I haven't checked the docs yet)

kartikmohta, 2021-02-09:
Looked at a simple example here: https://github.com/actions/starter-workflows/blob/main/ci/cmake.yml
Seems simple enough to copy the steps from `wercker.yaml` to this format.

tdinesh, 2021-02-09:
Ok, will give it a try.

tdinesh, 2021-02-09:
Addressed in #137

closed 2021-02-09 by tdinesh