KumarRobotics/kr_mav_control #138: Why do we use PositionCommand::constPtr in the interface?

versatran01, 2021-03-09:
I struggle to see the reason? Was it because of nodelet?

kartikmohta, 2021-03-10:
const because we don't want to allow changes outside.
Ptr because sometimes no output (nullptr) is desired.

If there are any applications which need non-const Ptr we can change.

versatran01, 2021-03-10:
When is no output desired?

I'm not questioning the const part, I'm just wondering whether we need a shared_ptr.

kartikmohta, 2021-03-10:
> When is no output desired?
> 

When there is a fancy tracker which directly wants to send the SO3Command and skip the position controller altogether.

> I'm not questioning the const part, I'm just wondering whether we need a shared_ptr.

shared_ptr is not really needed, could be a unique_ptr too but shared_ptr (that too the boost one) being the only one supported by nodelet for zero-copy was a decision in using it. boost::optional could also be used I guess.

closed 2021-03-11 by versatran01