KumarRobotics/kr_mav_control #147: mavros interface for trajectory tracker

shubham-shahh, 2022-04-07:
Hi,
I want to pass the SO3 command to the FCU via mavros, I checked the AttitudeTarget message, but that message only supports normlised thrust in z-axis, but SO3Command has forces in x, y and z axis. how can I convert SO3Command to Attitude target message

thanks

jessweakly, 2023-06-29:
This is an old thread so hopefully you have gotten the info you need, but I'm putting a response here in case someone has the same question. 

This should be handled already in kr_mavros_interface. Check out line 123 in particular. You will need to update the kf parameter based on your robot. https://github.com/KumarRobotics/kr_mav_control/blob/master/interfaces/kr_mavros_interface/src/so3cmd_to_mavros_nodelet.cpp

fcladera, 2024-02-16:
Thanks @jessweakly!!

closed 2024-02-16 by fcladera