KumarRobotics/kr_mav_control #157: How to tune the so3_control gains?

mzahana, 2023-09-22:
Hi.

I would like to know if there is a systematic method of tuning the so3_control gains, especially for actual drones.

How to choose an initial set of values, and how to tune them?

Thank you.

fcladera, 2024-02-16:
Tuning closed loop control systems is a topic in itself :D

Penn [has a class](https://platform.onlinelearning.upenn.edu/offering/robotics-aerial-robotics-a0Q2E00000JmMQzUAN) on aerial robotics where these topics are addressed.

As a general rule, once you get your mass and prop coefficients right, you send step inputs and evaluate the response of the system. [This wikipedia article](https://en.wikipedia.org/wiki/Proportional%E2%80%93integral%E2%80%93derivative_controller)  has some figures showing the effects of each parameter.

mzahana, 2024-02-16:
@fcladera Thanks for your reply. I agree it is a topic by itself with many research work, but mostly for linear controllers e.g. PID. In fact, I have some research work related to optimal control tuning (https://doi.org/10.3390/s22239240). However, it is designed for linear controllers. Whereas, the so3 is a nonlinear one.

jakewelde, 2024-02-18:
Usually, we tune these controllers by tuning the linearization, and then just use those gains for the full nonlinear controller. Practically speaking, this looks like tuning the gains for small initial errors. This is quite effective in practice. See the "Controller Gains" section of [this paper](https://ieeexplore-ieee-org.proxy.library.upenn.edu/document/8556372) for a discussion of a similar approach for a slightly different controller.